gepd/Deviot #245: annoyance, Serial slow to start after upload

tablatronix, 2018-07-01:
I cannot where I previously wrote this , I am creating a new issue.

It seems the serial console takes 2-3 seconds to start after upload because I miss any serial logging during that time, I have to add a delay to my sketches of at least 2 seconds or I miss output when it resets.

This seems to have changed at some point as it was not always an issue.

Any optimizations here would be worth it I think.

gepd, 2019-02-17:
Do you keep having this issue?

What S.O were you using?

tablatronix, 2019-02-17:
Its both osx and win

gepd, 2019-02-17:
and what board are you using? 
I'm trying with an esp32 and I'm not getting that lag.

Does it happen you with any sketch?

tablatronix, 2019-02-17:
Yup I have to use a 2000 delay in setup to see the first serial print etc

tablatronix, 2019-02-17:
Ill make a video

tablatronix, 2019-02-17:
https://youtu.be/lIDykeUlWrM

tablatronix, 2019-02-17:
I noticed there is some kinf percentage flashing at the bottom right footer, goes real quick when restarting serial, is it reloading the buffer or something ?

gepd, 2019-02-18:
I see I'll start to debug what can be causing that delay, I'm not sure if it's the same in my machine, but I'll test.

There is no buffer in the Deviot side. I think it could be the indexing feature from ST

tablatronix, 2019-02-18:
Hmm Ill disable all plugins and check again