gepd/Deviot #246: DEV Deviot not working anymore

tablatronix, 2018-07-18:
```
reloading plugin Deviot (Arduino IDE).start_listener
Traceback (most recent call last):
  File "C:\Users\...\Documents\projects\bin\sublimetext\sublime_plugin.py", line 116, in reload_plugin
    m = importlib.import_module(modulename)
  File "./python3.3/importlib/__init__.py", line 90, in import_module
  File "<frozen importlib._bootstrap>", line 1584, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1565, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1532, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 584, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1022, in load_module
  File "<frozen importlib._bootstrap>", line 1003, in load_module
  File "<frozen importlib._bootstrap>", line 560, in module_for_loader_wrapper
  File "<frozen importlib._bootstrap>", line 868, in _load_module
  File "<frozen importlib._bootstrap>", line 313, in _call_with_frames_removed
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\start_listener.py", line 17, in <module>
    from .commands import *
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\commands\__init__.py", line 58, in <module>
    from ..beginning.check_requirements import DeviotCheckRequirementsCommand
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\beginning\check_requirements.py", line 13, in <module>
    from .. import deviot
ImportError: cannot import name deviot
```

tablatronix, 2018-07-18:
I had to remove and reinstall it for some reason

tablatronix, 2018-07-18:
windows 10

gepd, 2018-07-18:
It should be solved restarting ST, ST won't see new files until restart.

tablatronix, 2018-07-18:
Yeah I restarted a few times, But it didn't work.

tablatronix, 2018-07-18:
Tested on osx other machine, same issue

gepd, 2018-07-18:
I'll try in other machine now

gepd, 2018-07-18:
I've replicated the problem, seems like package control is removing the file `deviot.py` and not replacing it, I'll see how to fix it

gepd, 2018-07-18:
Fixed with the last commit!

closed 2018-07-18 by gepd