gepd/Deviot #247: DEV boards issue

tablatronix, 2018-07-18:
osx, click boards does nothing, no errors logged

rebuild board list

```
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_rebuild_boards.py", line 13, in run
    PioBridge().save_boards_list_async()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/pio_bridge.py", line 32, in save_boards_list_async
    from ..beginning.pio_install import save_board_list
ImportError: No module named 'Deviot (Arduino IDE).beginning.pio_install'
```

gepd, 2018-07-18:
I'm sorry about that the last change broke that function but it's already fixed

gepd, 2018-07-19:
I'll add that to my to-do list