gepd/Deviot #252: dev osx fails on symlink libraries

tablatronix, 2018-07-27:
To work around another issue I added a symblink in libraries to a local library.
But it seems to break stuff
Maybe this can be ignored or fixed.

```
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
```

tablatronix, 2018-07-27:
and on rebuild lib

```
Exception in thread Thread-880:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 310, in save_installed_list
    self.quicked(out)
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 139, in quicked
    id = item['id']
KeyError: 'id'

```

tablatronix, 2018-07-27:
oops nope this is not the cause, It is something with the library.

tablatronix, 2018-07-27:
This must be something with this macbook still, my other one worked fine yesterday same, library.
suspect related to pathing

gepd, 2018-07-27:
I'll add the logger to debug all this problem this weekend, it sholdn't be happening and I don't think it's the src folder, I'll let you know

tablatronix, 2018-08-16:
Problem exists on windows also

```
Traceback (most recent call last):
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
Traceback (most recent call last):
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
Exception in thread Thread-46563:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\libraries.py", line 310, in save_installed_list
    self.quicked(out)
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\libraries.py", line 139, in quicked
    id = item['id']
KeyError: 'id'
```

gepd, 2018-08-17:
I've push a possible fix for this erros:
```python
Traceback (most recent call last):
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
Traceback (most recent call last):
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
```
The `KeyError: 'id'` it's triggered when you install a new library, do you know what library were you installing when this error appeared?
I can easily hide the error, but I'll like to know why the API is not sending an ID or I'm loosing it somewhere.

I need to replicate error