gepd/Deviot #253: dev remove library not working

tablatronix, 2018-07-27:
Seems like a token issue, similar to that fix you did with the library id arrays after you added the author.

remove library

```
[ Deviot 2.3.0.dev6 ] Library Manager
'mbed' is not recognized as an internal or external command,
operable program or batch file.

[07/27/18 09:29:39]
```

```
[ Deviot 2.3.0.dev6 ] Library Manager
'arduino' is not recognized as an internal or external command,
operable program or batch file.

[07/27/18 09:31:34]
```

tablatronix, 2018-07-27:
I also have "Select a Library" 3 times in my menu.

tablatronix, 2018-07-27:
I tried rebuild libraries

```
Exception in thread Thread-84:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\libraries.py", line 268, in remove_library
    lib_id = self.quick_list[selected][2]
IndexError: list index out of range

Exception in thread Thread-87:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\libraries.py", line 310, in save_installed_list
    self.quicked(out)
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\libraries\libraries.py", line 139, in quicked
    id = item['id']
KeyError: 'id'
```

gepd, 2018-07-27:
mmm I tried to replicated the first problem but I couldn't, can you paste this lines to debug the code: 

In this file and line: https://github.com/gepd/Deviot/blob/develop/platformio/command.py#L181

```python
print(env_path)
print(cmd)
```
Is PlatfomIO installed locally (by deviot) or are you using it globally?

After paste that lines, restarts ST and try to remove a library again, you should see new information in the console

tablatronix, 2019-01-12:
It looks there is a whole bunch of stuff that fails if a library was manually saved and as such does not have an id

[image]

closed 2021-06-22 by tablatronix