gepd/Deviot #254: possible cause of some issues on OSX

tablatronix, 2018-08-07:
```
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
Exception in thread Thread-1468:
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
Deviot (Arduino IDE).libraries.pyserial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 138, in receive
    inp_text = self.serial.read(buf_number)
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
Deviot (Arduino IDE).libraries.pyserial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

```

On osx, sometimes starting serial doesn't work, it gets stuck I have to toggle 3 times, then it creates a new group also. I logged this today.

tablatronix, 2018-08-07:
I think this happens when removing or adding devices

gepd, 2018-08-09:
Is this happent to you with all your devices?
what device are you using?

tablatronix, 2018-08-09:
I think both were cp2102

tablatronix, 2018-08-09:
ohh let me try not using my usbc dongle

tablatronix, 2018-08-16:
ok this issue is not just osx, happens on windows also.
The problem is the port stays open after closing the monitor , so it gets stuck in a inverted state

open monitor
close the tab its in
monitor port stays open, status bar stays green

(menu says start always)
try to start monitor again , and it stops, bar goes red

tablatronix, 2018-10-11:
this also causes serial monitor to get stuck on, or not stopped when uploading resulting it upload failures. So it seems there is some kind of loss of state of serial , where it gets inverted etc.

gepd, 2019-02-18:
I've made some tweaks in the way to handle the status bar color and you shouldn't problems now, however I wasn't able to reproduce the `read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)` error

Can you check if you keep having it?

tablatronix, 2019-09-24:
It looks like there is a bug in the serial state tracking..
I start serial monitor using deviot quick menu, it is running..
But menu thinks it is not

[image]

This happens now and then and seems to be related maybe to #292

tablatronix, 2019-10-01:
I think this all started when you added the serial popup every time starting serial.

It seems to break constantly , serial always says its running, menu says its not, cannot run it after ota, it says running but no data, I have to send to reactivate it, but send its always greyed out.. its very strange and annoying, I will have to try to din into serial.py and figure out wtf is going on or at least add some error handling, atm I am python copy paste level expertise.. lol

gepd, 2019-10-01:
I promise I'll check it this weekend!

tablatronix, 2019-10-01:
lol, if not give me some hints and I can hack at it.. or test it outside of deviot, if its a python command line call

tablatronix, 2019-10-05:
I just realized that pressing the shortcut for send, opens a dialog to pick serial port and then does nothing. If I pick send from menu it pops up the send input and works fine. This might be also breaking some state.

gepd, 2019-10-07:
can you check if the last commit solves this problem?

tablatronix, 2019-10-07:
Seems more stable, I will test more tomorrow

tablatronix, 2019-10-07:
Seems to have issues uploading now with serial running, as if serial is blocking it and not stopping and restarting or what it used to do

```
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266-Tone-Generator>
Compiling .pio/build/nodemcuv2/src/Demo_Tones.ino.cpp.o
Retrieving maximum program size .pio/build/nodemcuv2/firmware.elf
Checking size .pio/build/nodemcuv2/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  32.9% (used 26984 bytes from 81920 bytes)
PROGRAM: [===       ]  25.4% (used 264876 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/cu.SLAB_USBtoUART
Uploading .pio/build/nodemcuv2/firmware.bin
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP8266EX
Features: WiFi

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
========================== [FAILED] Took 9.92 seconds ==========================

[Sun Oct  6 21:24:38 2019]
```

tablatronix, 2019-10-07:
ALso new error, not sure if its related or actually new.

```
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 267, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 404, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
AttributeError: 'Command' object has no attribute 'run_command'
Backup saved to: /Users/shawn/.sublime/backups/Users/shawn/projects/microcontrollers/dev/libraries/ESP8266-Tone-Generator/examples/Demo_Tones/Demo_Tones-2019-10-06_21-23-27.ino
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 267, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 404, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
AttributeError: 'Command' object has no attribute 'run_command'
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 267, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 404, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
AttributeError: 'Command' object has no attribute 'run_command'
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 267, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 404, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
AttributeError: 'Command' object has no attribute 'run_command'
```

tablatronix, 2019-10-13:
Yup cannot upload when serial monitor is in use

```
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Traceback (most recent call last):
Chip is ESP8266EX
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 2959, in <module>
    _main()
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 2952, in _main
    main()
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 2665, in main
    print("Features: %s" % ", ".join(esp.get_chip_features()))
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 974, in get_chip_features
    if self.get_chip_description() == "ESP8285":
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 968, in get_chip_description
    efuses = self.get_efuses()
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 965, in get_efuses
    self.read_reg(0x3ff00050))
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 475, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 332, in command
    p = self.read()
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 277, in read
    return next(self._slip_reader)
  File "/Users/shawn/.platformio/packages/tool-esptoolpy/esptool.py", line 1873, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
*** [upload] Error 1
�========================== [FAILED] Took 4.49 seconds ==========================

Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  IGNORED
nodemcuv2            FAILED    00:00:04.486
esp12e               IGNORED
==================== 1 failed, 0 succeeded in 00:00:04.486 ====================

```

gepd, 2019-10-17:
I've revert a commit who should be causing this errors, can you test it now?

tablatronix, 2020-03-04:
```
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 493, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 141, in receive
    inp_text = self.serial.read(buf_number)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
Deviot (Arduino IDE).libraries.pyserial.serialutil.SerialException: read failed: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 144, in receive
    toggle_serial_monitor(self.port)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 304, in toggle_serial_monitor
    message.initial_text("_deviot_{0}", version)
NameError: global name 'version' is not defined

Backup saved to: /Users/shawn/.sublime/backups/Users/shawn/projects/microcontrollers/dev/libraries/IoAbstraction/examples/interruptSwitchEncoder8574/interruptSwitchEncoder8574-2020-03-03_21-39-45.ino
reloading settings Packages/User/deviot.sublime-settings
```