gepd/Deviot #257: Board restart twice when uploading while monitoring

OoDeLally, 2018-08-17:
### Operating system
Linux Mint + ST3

### Board used
Arduino Uno

### Description of problem
When hitting the `Upload` button, it recompiles, uploads and reset the board. This is great.
If we do that while monitoring is enabled, the board is restarted twice: 
1- The program runs for around ~0.5sec and is immediately aborted after this short delay.
2- Then immediately reset again. This time the program runs for good.

### Error(s) in ST Console? (View > Show Console)
The sublime console displays
```reloading settings Packages/User/deviot.sublime-settings``` at the moment i press the `Upload` command```, and then displays a second time
```reloading settings Packages/User/deviot.sublime-settings```
when the fake (first) run starts.
It doesnt print anything when the real run (second) starts.

### Steps to Reproduce
Do as describe in the problem description. This works for any program.

### Actual Results
It seems that the board is reset twice at a ~0.5sec interval.

### Expected Results
The board should be reset only once after the upload.

### Additional info
It only happens when the monitoring (`Monitor Serial -> Start monitor`) is enable.

gepd, 2018-08-19:
`reloading settings Packages/User/deviot.sublime-settings` doesn't mean the board is reset twice, the reason is when you have the Serial Monitor running, the plugin stores that state and save it in a file, each time that file is writed ST log that change in the console (`reloading settings Packages/User/deviot.sublime-settings`) There are others changes happening in background so sometimes will see it more than twice but there is nothing to worry about it

OoDeLally, 2018-09-13:
I see. This was just an observation, and I did not know whether it was correlated. Thanks for the info.
The problem is still there. Do you have any insight?

gepd, 2019-02-17:
I'll close this as it's not a bug

closed 2019-02-17 by gepd