gepd/Deviot #258: Monitor crashes when upload command is triggered when focusing the monitor or Deviot console

OoDeLally, 2018-08-17:
### Operating system
Linux Mint

### Board used
Arduino Uno

### Description of problem
When uploading while monitoring with the `Alt-E` command, the monitor crashes and stops if the current focused window is either the Deviot console buffer, or the monitor buffer.

### Error(s) in ST Console? (View > Show Console)

```
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/home/pascal/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 49, in start_upload
    self.add_option('lib_extra_dirs', append=True)
  File "/home/pascal/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 295, in add_option
    config.read(ini_path)
  File "/home/pascal/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/readconfig/__init__.py", line 99, in read
    if(not path.exists(filepath)):
  File "./python3.3/genericpath.py", line 18, in exists
TypeError: can't specify None for path argument

top level value must be an array
```

### Steps to Reproduce
1- Enable monitoring (Alt-M).
2- Focus the monitor or Deviot buffer.
3- Hit the Upload command (Alt-E).

### Additional info

It seems very related to this issue https://github.com/gepd/Deviot/issues/209#issuecomment-364934377
But I could not figure out if this issue has been solved for this guy.

Thank you!

OoDeLally, 2018-09-13:
Yeah it solves it. Now it shows `is not an IoT File`, which is a lot better than crashing.
I understand you need to designate which project to upload, and if you dont focus the right buffer, Deviot cannot know which. However it is rare that we work on several projects at the same time.
Is it possible to simply uploads the last uploaded project, if no valid buffer is on focus? Of course it is a luxury feature, but it would be nice :+1:

tablatronix, 2018-10-09:
You can use the freeze sketch in the compile/upload options menu for now.

gepd, 2019-02-17:
Let me know if [freeze sketch](https://github.com/gepd/Deviot/wiki/Deviot-Menu#freeze-sketch) is enough  for you.

Re open it if you need something extra

closed 2019-02-17 by gepd

tablatronix, 2019-02-17:
I just happen to notice that if compiling/uploading from serial does not use freeze , it still says not an iot file.