gepd/Deviot #263: Is there any way to initialize platformio.ini with user defined options?

victorpraxedes, 2018-09-06:
I'm building a code repository which will have several third party libraries for learning purposes, the folder looks like this:

root
│
├───libs
│        ├───sensor_lib
│        └───display_lib
│ 
├───some_project
│        ├───lib
│        └───src
│        └───platformio.ini
│
├───other_project
          ├───lib
          └───src
          └───platformio.ini

What i'm looking for is to add ```lib_extra_dirs = ..\libs``` to the automatically generated ```platformio.ini``` when i first hit ```Deviot > Compile``` so I don't have to manually add it every time I create a new project.

gepd, 2018-09-08:
You can go to `Preferences > Packages Settings > Deviot > Settings` and adds `lib_extra_dirs` to the settings with your libraries sapared by comma

Your file should looks like this:

```json
{
	"lib_extra_dirs": "pathtolibrary1, pathtolibrary2"
}
```
With this, each time you compile your sketch the extra libraries will be add to `platformio.ini`

pxds, 2018-09-10:
I did use this option, but the path it used was absolute:
```
{
	"lib_extra_dirs": "C:\projects\foobar\libs"
}
```
Is Deviot smart enough to know where to look when i clone this into another machine?

victorpraxedes, 2018-09-10:
I did use this option, but the path it used was absolute:
```
{
	"lib_extra_dirs": "C:\projects\foobar\libs"
}
```
Is Deviot smart enough to know where to look when i clone this into another machine?

gepd, 2018-09-12:
at this moment Deviot can't use all library options available in PlatformIO as you can see here https://docs.platformio.org/en/latest/projectconf/section_env_library.html

So what you can do is to have a copy of the `platformio.ini` with your default settings and use it each time you create a new proyect

victorpraxedes, 2018-09-12:
Ok, that was my workaround already :) Thanks for the answers.