gepd/Deviot #264: DEV linter errors persist

tablatronix, 2018-10-09:
It seems that linter context errors, stay in a buffer even after recompile.
If there is an error, previous errors show back up, even after closing them.

### reproduce

make syntax error
compile bad
close red `error`s

compile successfully
make different syntax error
compile bad with a different error

same errors show back up from first time

tablatronix, 2018-10-31:
any ideas how to clear these out without restarting?

[image]

tablatronix, 2018-11-06:
`self.errs_by_file[file]` is never cleared, errors just keep appending.
not sure where to put a clear for compile command only? or how to get self.file in command, for now I am just emptying the entire thing on command.

It seems not very efficient to be processing errors and phantoms on data constantly, not sure if there is a way to do it once at end, probably doesn't matter seemed like excessive processing.

gepd, 2018-11-16:
I've pushed a temporal fix for this issue, let me know if it's better

tablatronix, 2018-11-16:
No, also I did not see any commits related to this fix, you sure it pushed?

tablatronix, 2018-11-16:
if you mean this?
https://github.com/gepd/Deviot/commit/5560fb3633c7c66f6332ee18ef12530e1e60ef37

thats just the setting toggle

tablatronix, 2019-01-12:
works great, what do you think about clearing errors after successful compile, if you do not X them they just stick around