gepd/Deviot #267: dev new error suddenly

tablatronix, 2018-12-10:
I was using it, and all of a sudden this happened anytime I run anything...

osx

Any ideas?

```
[ Deviot 2.3.0.dev8 ] Starting...
Traceback (most recent call last):
  File "/usr/local/bin/platformio", line 9, in <module>
    load_entry_point('platformio==3.6.0a10', 'console_scripts', 'platformio')()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 565, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 2696, in load_entry_point
    raise ImportError("Entry point %r not found" % ((group, name),))
ImportError: Entry point ('console_scripts', 'platformio') not found

[Mon Dec 10 10:03:31 2018]
```

tablatronix, 2018-12-10:
I removed user/penv

now I get

```
Exception in thread Thread-24:
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 71, in start_upload
    if(not self.check_auth_ota()):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/project_check.py", line 289, in check_auth_ota
    platform = self.get_platform()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 115, in get_platform
    boards = boards_file.read_json()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
```

tablatronix, 2018-12-10:
run platformio from terminal I get
```pkg_resources.DistributionNotFound: The 'platformio==3.6.0a10' distribution was not found and is required by the application```

tablatronix, 2018-12-10:
uncheck use development version, says installing stable, there was an error

gepd, 2018-12-11:
In which O.S it's given you this error?

tablatronix, 2018-12-11:
sorry it was not very obvious in first post

Osx mojave

gepd, 2018-12-13:
The fastest solution would be install platformio from `Homebrew` and then remove` user / Deviot` to deviot recognize it

tablatronix, 2018-12-13:
Yeah its bizarre, platformio said create a bug report,  no idea where that alpha version that doesnt exists is coming from, going to wipe it, what deviot files do I need to restore all ny settings?

gepd, 2018-12-13:
If you were using PlatformIO installed from Deviot, all files are located in `User/Deviot` 
The `penv` folder stores PIO, and the rest of the files are for configuration. If you remove `User/Deviot` and restart ST, Deviot will install the stable version of PIO again

tablatronix, 2018-12-13:
Yeah I tried all the usual stuff already, deviot didnt do anything when I remove it and start sublime. 

Isnt my saved settings in there also ?

gepd, 2018-12-14:
Isn't you should remove `User/Deviot` or the `deviot.ini` file in the same folder

tablatronix, 2018-12-20:
Yeah something must have broken again, that did nothing. I removed package and reinstalled it.

and it still gives an error, and I still have no user/Deviot folder, is never gets created..

```
Setup done
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_libraries_examples.py", line 6, in run
    QuickMenu().quick_libraries()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 367, in quick_libraries
    platform = self.get_platform()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 115, in get_platform
    boards = boards_file.read_json()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_select_boards.py", line 8, in run
    items = Quick.boards_list()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 88, in boards_list
    boards = boards_file.read_json()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
```

tablatronix, 2018-12-20:
ok I reinstalled and the boards.json file has stderr out in it

boards.json

```
Traceback (most recent call last):
  File "/usr/local/bin/platformio", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3095, in <module>
    @_call_aside
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3081, in _call_aside
    f(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3108, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 658, in _build_master
    ws.require(__requires__)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 959, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 846, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==3.6.0a10' distribution was not found and is required by the application

```

tablatronix, 2018-12-20:
So it looks like a pio issue that created this

https://community.platformio.org/t/vsc-the-platformio-3-5-0a17-distribution-was-not-found-and-is-required-by-the-application/2855

tablatronix, 2018-12-20:
I would still love to fix this darn macbook, so it stops using global python and pio, I still have no idea wtf is wrong with it

tablatronix, 2018-12-20:
This is my path for reference maybe someone will know something, Could it be a problem with sublime ? I have not reinstalled it

```
/usr/local/bin
/usr/bin
/bin
/usr/sbin
/sbin
/Users/.../Library/Python/2.7/bin
/Applications/Wireshark.app/Contents/MacOS
```

tablatronix, 2018-12-20:
ok I think I am back in business, I removed platformio, pio from  /usr/local/bin
and removed deviot.ini, and removed junk from boards.json and restarted

closed 2018-12-20 by tablatronix