gepd/Deviot #271: DEV BUG setting extra library folder

tablatronix, 2019-01-04:
If I try to change my extra lib folder via the gui

```
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/api/deviot.py", line 654, in <lambda>
    path, callback, key, paths_list, index)), 0)
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/api/deviot.py", line 613, in folder_explorer
    prev = path.dirname(path)
AttributeError: 'str' object has no attribute 'dirname'
```

tablatronix, 2019-01-12:
```
path
/Users/shawn/projects/microcontrollers/dev/platformio/build
```

tablatronix, 2019-01-12:
using conflicting var names, scopes in folder_explorer

`        prev = path.dirname(path)`

tablatronix, 2019-01-12:
changed all folder_explorer `path`(str) to `pathstr` fixes issue since path is os.path

gepd, 2019-02-14:
Fixed with https://github.com/gepd/Deviot/commit/0042e51b687572eb134a60407e3d87ac257af79b

closed 2019-02-14 by gepd