gepd/Deviot #272: DEV BUG

tablatronix, 2019-01-10:


tablatronix, 2019-01-10:
if you have a comment in your platformio.ini upload speed overwrite gets corrupt ( newline added )

platformio.exception.InvalidProjectConf: Invalid `platformio.ini`, project configuration file: 'File contains parsing errors: /Users/.../projects/microcontrollers/dev/build/OnDemandConfigPortal/platformio.ini
	[line 35]: '/Users/alverson/projects/microcontrollers/dev/libraries\n''

```
[env:nodemcuv2_dev]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
upload_speed = 
921600
; build_flags =-DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -Wall -Wextra
```

tablatronix, 2019-01-10:
hmm this might only occur is ota and `upload_flags = --auth=1234` is present in the file..
odd

gepd, 2019-02-17:
I've updated to `dev9` https://github.com/gepd/Deviot/commit/5f6e399cc88c5e2bb4e3b3d5194f13b0455a3256 should solve this problem.

If this isn't can you give a detailed example to reproduce the problem?

tablatronix, 2019-02-17:
Ill test bit see the build line above I have ; commented out, when deviot touches the file it puts a newline as above

tablatronix, 2020-03-18:
Deviot completely corrupts platformio.ini files
No idea what it is doing seems to remove comments and messes up newlines.

Every-time you try to compile so there is no way around it failing.

[image]

tablatronix, 2020-05-12:
#311

tablatronix, 2020-07-12:
Let me see if I am indeed using this branch, will post back!

tablatronix, 2020-07-12:
[ Deviot 2.3.0.dev12 ] Starting...

So it is something else an this did not fix it.

tablatronix, 2020-07-12:
lol well i was trying to create reproduction steps but I cannot install new libraries. 

```
Exception in thread Thread-45:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 99, in download_list
    response = urlopen(req)
  File "./python3.3/urllib/request.py", line 156, in urlopen
  File "./python3.3/urllib/request.py", line 475, in open
  File "./python3.3/urllib/request.py", line 587, in http_response
  File "./python3.3/urllib/request.py", line 513, in error
  File "./python3.3/urllib/request.py", line 447, in _call_chain
  File "./python3.3/urllib/request.py", line 595, in http_error_default
urllib.error.HTTPError: HTTP Error 429: Too Many Requests
```
Let me wait and see if it resets soon...

tablatronix, 2020-07-12:
Ok

- Installl aircookie/wled library
- Open wled00/wled00.ino
- Select enviroment, (nodemcu v2, any)
- Compile
- Fail
```PHP
[ Deviot 2.3.0.dev12 ] Starting...
Error: Invalid '/Users/shawn/projects/microcontrollers/dev/libraries/WLED/platformio.ini' (project configuration file): 'File contains parsing errors: /Users/shawn/projects/microcontrollers/dev/libraries/WLED/platformio.ini
	[line 168]: 'nodemcuv2\n'
	[line 170]: '${common.platform_wled_default}\n'
	[line 178]: 'esp01\n'
	[line 180]: '${common.platform_wled_default}\n'
	[line 183]: '${common.build_flags_esp8266} -D WLED_DISABLE_OTA -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK\n'
	[line 188]: 'esp01_1m\n'
	[line 190]: '${common.platform_wled_default}\n'
	[line 196]: 'esp01_1m\n'
	[line 198]: '${common.platform_wled_default}\n'
	[line 204]: 'esp07\n'
	[line 206]: '${common.platform_wled_default}\n'
	[line 213]: '${common.platform_wled_default}\n'
	[line 215]: '921600\n'
	[line 221]: 'd1_mini\n'
	[line 223]: '${common.platform_wled_default}\n'
	[line 229]: 'd1_mini\n'
	[line 231]: '${common.platform_wled_default}\n'
	[line 238]: 'espressif32@1.11.2\n'
	[line 240]: '${common.build_flags_esp32}\n'
	[line 247]: 'esp8285\n'
	[line 249]: '${common.platform_latest}\n'
	[line 255]: 'esp8285\n'
	[line 257]: '${common.platform_latest}\n'
	[line 263]: 'esp8285\n'
	[line 265]: '${common.platform_latest}\n'
	[line 271]: 'd1_mini\n'
	[line 273]: '${common.platform_latest}\n'
	[line 284]: 'debug\n'
	[line 286]: '${common.platform_wled_default}\n'
	[line 295]: '"10.10.1.27"\n'
	[line 297]: '${common.platform_wled_default}\n'
	[line 307]: 'd1_mini\n'
	[line 309]: '${common.platform_latest}\n'
	[line 315]: 'd1_mini\n'
	[line 317]: '${common.platform_latest}\n'
	[line 323]: 'd1_mini\n'
	[line 325]: '${common.platform_latest}\n'
	[line 331]: 'd1_mini\n'
	[line 333]: '${common.platform_latest}\n'
	[line 340]: 'espressif32@1.11.2\n'
	[line 353]: '460800\n''

[Sun Jul 12 11:16:52 2020]
```

- Diff platformio.ini and see the bad newlines etc.

tablatronix, 2020-07-12:
Comment counting is wrong..

```pl
                        if(len(comcount) == 0):
                            values = '\n' + values ; <-- NEWLINES after platform = 
```

tablatronix, 2020-07-12:
This is the diff of corrupt files, when a new line gets inserted wrong

```diff
--- /Users/shawn/projects/microcontrollers/dev/libraries/WLED/platformio.ini
+++ (clipboard)
@@ -2,25 +2,20 @@
 ; Please visit documentation: https://docs.platformio.org/page/projectconf.html
 
 [platformio]
-src_dir  = ./wled00
+src_dir = ./wled00
 data_dir = ./wled00/data
-lib_dir  = ./wled00/src
+lib_dir = ./wled00/src
 build_cache_dir = ~/.buildcache
-extra_configs = 
-  platformio_override.ini
-
+extra_configs =   platformio_override.ini
 # ------------------------------------------------------------------------------
 # ENVIRONMENTS
 #
 # Please uncomment one of the lines below to select your board(s)
 # ------------------------------------------------------------------------------
-
 # Travis CI binaries
 default_envs = travis_esp8266, esp01, esp01_1m_ota, travis_esp32
-
 # Release binaries
 ; default_envs = nodemcuv2, esp01, esp01_1m_ota, esp01_1m_full, esp32dev, custom_WS2801, custom_APA102, custom_LEDPIN_16, custom_LEDPIN_4, custom32_LEDPIN_16
-
 # Single binaries (uncomment your board)
 ; default_envs = nodemcuv2
 ; default_envs = esp01
@@ -40,23 +35,11 @@
 ; default_envs = wemos_shield_esp32
 ; default_envs = m5atom
 
+
+
+
+
 [common]
-# ------------------------------------------------------------------------------
-# PLATFORM:
-#   !! DO NOT confuse platformio's ESP8266 development platform with Arduino core for ESP8266
-#
-#   arduino core 2.3.0 = platformIO 1.5.0
-#   arduino core 2.4.0 = platformIO 1.6.0
-#   arduino core 2.4.1 = platformIO 1.7.3
-#   arduino core 2.4.2 = platformIO 1.8.0
-#   arduino core 2.5.0 = platformIO 2.0.4
-#   arduino core 2.5.1 = platformIO 2.1.1
-#   arduino core 2.5.2 = platformIO 2.2.3
-#   arduino core 2.6.1 = platformIO 2.3.0
-#   arduino core 2.6.2 = platformIO 2.3.1
-#   arduino core 2.6.3 = platformIO 2.3.2
-#   arduino core 2.7.0 = platformIO 2.5.0
-# ------------------------------------------------------------------------------
 arduino_core_2_3_0 = espressif8266@1.5.0
 arduino_core_2_4_0 = espressif8266@1.6.0
 arduino_core_2_4_1 = espressif8266@1.7.3
@@ -68,24 +51,20 @@
 arduino_core_2_6_2 = espressif8266@2.3.1
 arduino_core_2_6_3 = espressif8266@2.3.3
 arduino_core_2_7_1 = espressif8266@2.5.1
-
 # Development platforms
 arduino_core_develop = https://github.com/platformio/platform-espressif8266#develop
 arduino_core_git = https://github.com/platformio/platform-espressif8266#feature/stage
[... 304 lines cut ...]
 upload_speed = 460800
-build_flags = ${common.build_flags_esp32} -D LEDPIN=16 -D RLYPIN=19 -D BTNPIN=17
+build_flags =
+${common.build_flags_esp32} -D LEDPIN=16 -D RLYPIN=19 -D BTNPIN=17
 lib_ignore =
   ESPAsyncTCP
   ESPAsyncUDP
@@ -334,14 +341,14 @@
 [env:m5atom]
 board = esp32dev
 build_flags = ${common.build_flags_esp32} -D LEDPIN=27 -D BTNPIN=39
-lib_ignore = 
+lib_ignore =
 	ESPAsyncTCP
 	ESPAsyncUDP
 platform = espressif32@1.11.2
-
 # ------------------------------------------------------------------------------
 # travis test board configurations
 # ------------------------------------------------------------------------------
+
 
 [env:travis_esp8266]
 extends = env:d1_mini
```

gepd, 2020-07-13:
> lol well i was trying to create reproduction steps but I cannot install new libraries, ugh rate limited somehow...
> 
> ```
> Exception in thread Thread-45:
> Traceback (most recent call last):
>   File "./python3.3/threading.py", line 901, in _bootstrap_inner
>   File "./python3.3/threading.py", line 858, in run
>   File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 99, in download_list
>     response = urlopen(req)
>   File "./python3.3/urllib/request.py", line 156, in urlopen
>   File "./python3.3/urllib/request.py", line 475, in open
>   File "./python3.3/urllib/request.py", line 587, in http_response
>   File "./python3.3/urllib/request.py", line 513, in error
>   File "./python3.3/urllib/request.py", line 447, in _call_chain
>   File "./python3.3/urllib/request.py", line 595, in http_error_default
> urllib.error.HTTPError: HTTP Error 429: Too Many Requests
> ```
> 
> Let me wait and see if it resets soon...

It isn't a rate limit look https://github.com/gepd/Deviot/issues/313

I think I forgot to apply some changes to the dev branch, I'll do it in a few hours

> Comment counting eval handling is wrong
> 
> readconfig write()
> 
> ```perl
>                         if(len(comcount) == 0):
>                             values = '\n' + values ; <-- NEWLINES after platform = 
> ```
> 
> This seems wrong, If I remove the \n it works

I'll check this again, It has been a while since I saw this code for the last time, it shouldn't be so hard to fix

tablatronix, 2020-12-24:
This has been working pretty good, but there seems to be 1 more bug, debug-flags sections are being removed