gepd/Deviot #273: OTA auth fail if using device selection, must use ip manual

tablatronix, 2019-01-13:
I have to manually enter the ip to get ota working, if I select the ip in devices it does not add an auth token and fails, it also does not prompts for password etc.

[image]

tablatronix, 2019-01-13:
```
Use manually specified: no
esptool -vv -cd nodemcu -cb 921600 -cp "no" -cf .pioenvs/nodemcuv2/firmware.bin
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 921600
setting port from /dev/tty.usbserial to no
espcomm_upload_file
espcomm_upload_mem
opening port no at 921600
error: cannot access no
```

tablatronix, 2019-01-13:
if I use device mdns selection, it also fails if you enter auth key

```
Use manually specified: no
esptool --auth=1234 -vv -cd nodemcu -cb 921600 -cp "no" -cf .pioenvs/nodemcuv2/firmware.bin
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
error: Invalid argument or value after --auth=1234 (argument #2)
*** [upload] Error 2
========================== [ERROR] Took 3.62 seconds ==========================

================================== [SUMMARY] ==================================
Environment nodemcuv2	[ERROR]
Environment esp32dev 	[SKIP]
========================== [ERROR] Took 3.62 seconds ==========================

[Sun Jan 13 09:53:17 2019]
```

gepd, 2019-02-14:
Fixed with https://github.com/gepd/Deviot/commit/b63913d5b948959d6bc7a0cd2b54199cb62a91ec
Can you check it?

closed 2019-02-17 by gepd