gepd/Deviot #277: exception serial readiness

tablatronix, 2019-02-16:
This is probably an edge case, but posting it here so it can be handled properly.

```C++
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
Deviot (Arduino IDE).libraries.pyserial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 139, in receive
    inp_text = self.serial.read(buf_number)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
Deviot (Arduino IDE).libraries.pyserial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

```

gepd, 2019-02-16:
Does that happened when you tried to open the monitor serial?

tablatronix, 2019-02-16:
Yes, I will try to reproduce and see what caused, i had to restart st to fix it.

gepd, 2019-02-17:
This is the same as https://github.com/gepd/Deviot/issues/254
I'll close this one

closed 2019-02-17 by gepd