gepd/Deviot #280: Dev new issue?

tablatronix, 2019-02-19:
I just noticed this with one of the new commit I think, when closing the serial tab , the group gets closed, kind of annoying. But I noticed open serial no longer creates additional groups which is nice

gepd, 2019-02-19:
Yes, now it closes the group (only if it's empty) It should be had working that way from the beginning. I can add an option to select if you want that behavior or leave group open

 To avoid create an aditional group you have to add the `monitor_direction` parameter in the setting file (`Preferences > Package Settings > Deviot > Settings`)

tablatronix, 2019-02-20:
ill test it out, I usually have a second group always open for it,  I will see if this works now decently or desirable.

tablatronix, 2019-02-20:
ok works pretty nicely, I would need an option to not close though , since I usually have groups open and open multiple serials in different ones so my layout does not get messed up and resized etc, and would like to leave my groups open.

Sidenote, 
I always use multiple groups, I  goto a group, and then start serial there ( which is what I usually do since I use multiple serials in multiple group panes.) A interesting feature would be to open in monitor_direction but use the group if there is one already there. So if you open a serial in side a group it doesn't open another one. Now I have no idea if that's possible or what should be the main or first group, most likely not. Maybe if more than 1 group exits, open is the second one.. not sure a good balance but it would save having to focus an empty group when self mode is set.

```
monitor_direction: right
auto_open: false
auto_open_sngle: true;  auto open only if single layout
auto_close: false
```

gepd, 2019-02-20:
Do you mean; if there is only one group, open it in the same group `auto_open: false` and if you have more than one group, open it always in the second one?

What would be the use case with `auto_open_single`?

tablatronix, 2019-02-20:
hmm

If layout is single ( only one group ), then auto open in monitor_direction(if not self of course), aka create a new group

If there is more than one group do not create a new group, open where focus, or if possible open in group #2 or monitor_direction? ( no idea how these are identify-able ) can always just open in self I suppose.

gepd, 2019-02-20:
The self property will cover two cases

1. If you don't want to create a new group
2. Open where focus is (multiples groups)

To open it in a specific group isn't a problem, I can know how many groups the windows have and if they're empty or not, so I'm thinking to implement this in a flexible way.

I can add `window_position` option who can work this way:

By default the value of this option will be `1` which means, if `monitor_direction` is left or right, it will always open the windows in a new group (first one or last one), 

If the value is `>= 2` it will open the window in that position. 
If you have currently two groups and the value of `window_position` is 3, it will open it in the last group It won't create a new one

What do you think of that?

tablatronix, 2019-02-20:
Thats sounds very cool

gepd, 2019-02-20:
Implemented with the last two commits, let me know what you think