gepd/Deviot #281: DEV upload failing

tablatronix, 2019-02-21:
```PHP
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 52, in start_upload
    self.add_option('lib_extra_dirs', append=True)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 325, in add_option
    option = self.init_option + ', ' + option
TypeError: unsupported operand type(s) for +: 'int' and 'str'
```

tablatronix, 2019-02-21:
It was working fine then suddenly started doing this, cannot upload

tablatronix, 2019-02-21:
ahh I did a untouch, and it messed up the ini file.

```
upload_speed = 921600
lib_extra_dirs = 921600 <----- whaaaa
```

gepd, 2019-02-21:
The last commit should fix it

tablatronix, 2019-02-21:
ok Ill check, I was testing always compiling issues, was seeing alot of compiling ( I am having upload issues atm so testing alot of reuplods )