gepd/Deviot #286: DEV: upload always compiles..

tablatronix, 2019-03-03:
Not sure if you updated something but, I am having compilations every time I upload now.
toggling untouch platformio made no difference.

OSX

gepd, 2019-03-05:
You make sure untouch is always unselected to avoid re-compilation.

In this case it was happening because when you override the baudrate it's added when you press `upload`, so it was modifing the `platformio.ini` and it made re-compile the sketch. 

I've solve that with: https://github.com/gepd/Deviot/commit/5e8903ecf933dbfa5d002b7f4d6e683ec8c30865

tablatronix, 2019-03-05:
I thought it was the opposite.. lol

gepd, 2019-03-05:
Yes, I think some times it's a little confuse, it should have a better name.

Untouch will modify `platformio.ini` before and after compile/upload, to keep the file 'clean'. Everytime `platformio.ini` is modified, the sketch is re-compiled (even if it have the same information)

tablatronix, 2019-03-06:
Is that a deviot doing that or platformio? Curious

gepd, 2019-03-06:
PlatformIO is who decide when re-compile, after I added the untouch feature, I realized platformIO checks, I guess, the last time the `platformio.ini` was modified instead of the content of the file. As some users doesn't want extra information in its file (like library, baudrate, etc) I had to add the untouch option, who adds the options/flags before compile/upload and removes it when it finish, but with price to compile each time.

Without untouch I try to edit the file as little as possible

tablatronix, 2019-03-06:
Do you think it is related to this?
https://github.com/platformio/platformio-core/issues/2109

gepd, 2019-03-06:
It isn't because I'm not using the `ci` and `--keep-build-dir` flag, but I should do it, I'm seeing it can solve some problems. I'll do some test to see if it's a better option for the untouch and re-compile problems

Thanks for the info