gepd/Deviot #289: Feature Request - Migrate To Python 3 - Python 2.7 Near EOL

Sn3akyP3t3, 2019-05-09:
I would like for this project to succeed and progress.  Looks like Python 2.7 is end of life as of 2020 which is rapidly approaching.

"Being the last of the 2.x series, 2.7 will have an extended period of maintenance. Specifically, 2.7 will receive bugfix support until January 1, 2020. After the last release, 2.7 will receive no support.

Planned future release dates:

    2.7.17 mid 2019
    2.7.18 January 2020"

Then after that... darkness.

Source:
https://www.python.org/dev/peps/pep-0373/

Additionally, At the moment I'm unwilling to install this IDE due to the fact that it demands Python 2.7 which I don't want installed on my system removing or overriding my existing version with the environment variables.

gepd, 2019-05-31:
Thanks for the info @Sn3akyP3t3 currently deviot is supporting python 3 but PlatformIO is still in development, although you can use the development version without problem.

Check [this link](https://docs.platformio.org/en/latest/userguide/cmd_upgrade.html#id4) to install the dev branch

alexws54tk, 2019-06-14:
> I would like for this project to succeed and progress. Looks like Python 2.7 is end of life as of 2020 which is rapidly approaching.
>
> "Being the last of the 2.x series, 2.7 will have an extended [...]

Support for Windows XP ended April 8, 2014, but it does not prevent a half of the Planet to continue to use it.
Support for Windows 7 is coming to an end on January 14, 2020, but something not all rushed to be updated to a W10.

@Sn3akyP3t3, „No support“ != „End of Life”.

Use Debian, bro — they **very slowly** make changes in software versions.

abhiTronix, 2019-09-09:
@alexws54tk But what happens if there's a security issue found after January 1, 2020, do users that are affected have the motivation to pay for a fix? Nope. At some point, you have let go old stuff, or become an antique bank or something.