gepd/Deviot #290: Login into platformio account from deviot

krish2487, 2019-05-24:
### Operating system : 
Macos Mojave

### Board used: 
NA

### Description of problem :
How to login into platformio account using deviot.

### Error(s) in ST Console : 
NA

### Steps to Reproduce : 
1. Menu > Deviot > Show Terminal
2. Enter "pio account login"
3. Enter email address

### Actual Results : 
Receive invalid command error

### Expected Results:
I expect to be able to enter my email address and password and login into the pio account.

### Additional info
Screenshots of the issue.

[image]
[image]
[image]
[image]
[image]

gepd, 2019-05-31:
I'll improve this when you use only `platformio account login`, for now you do `platformio account login -u email@email.com -p password `

krish2487, 2019-06-01:
@gepd Thank you the suggestion. unfortunately it does not seem to work.
This is what I get when I try logging in using the method above.

[image]

gepd, 2019-06-02:
I had the same problem, what I did was install the developer version of platformio:

1. uninstall existing version
`pip uninstall platformio`

2. install the latest development version of PlatformIO
`pip install -U https://github.com/platformio/platformio-core/archive/develop.zip`

Currently I'm using `PlatformIO, version 4.0.0a16` you can check this with `platformio --version`

krish2487, 2019-06-03:
That worked!! :-)
Thank you for the help!!!

closed 2019-08-26 by krish2487