gepd/Deviot #291: Deviot freezes, saying 'processing' when trying to upload to board.

barnabasjohnthomas, 2019-07-14:
Before report an issue, please make sure it hasn't been reported before.

If you need information about an option or feature make sure to first look into the [Deviot Wiki](https://github.com/gepd/Deviot/wiki)

### Operating system
macOS High Sierra

### Board used
Arduino (Genuino) Uno

### Description of problem
Whenever I try to upload, Deviot just shows 'processing' in the status bar with the equal sign between the two square brackets.

### Error(s) in ST Console? (View > Show Console)
The Console doesn't show an error.

### Steps to Reproduce
Try to upload a sketch to the board.

### Actual Results
Doesn't upload.

### Expected Results
It uploads?

### Additional info
It works just fine with the Arduino IDE.

*You can edit the text above according to your need*

gepd, 2019-07-14:
Can you install the developer version and tell if you keed having the same behavior? 
https://github.com/gepd/Deviot/wiki/Setup#developer

barnabasjohnthomas, 2019-07-15:
Solved it… thank you! You’re the best my man :)

closed 2019-07-15 by barnabasjohnthomas