gepd/Deviot #293: Broken deviot on macos catalina

krish2487, 2019-08-22:
### MacOS catalina  10.15 Beta (19A536g)

### ESP32 hornbill board - Arduino Framework

### Description
Upgrading to catalina breaks the platformio build process from within sublime text.
The code does not compile irrespective of the platform used and gives an error

### Error(s) in ST Console? (View > Show Console)
```
[ Deviot 2.2.1 ] Starting...
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/platformio/__main__.py", line 19, in <module>
    import click
  File "/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/click/__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/click/core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/click/types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/lib/python2.7/site-packages/click/_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

[Thu Aug 22 09:52:55 2019]

```
### Steps to Reproduce
Upgrade to catalina and try to compile any test firmware for any platform

### Additional info
Running platformio manually compiles the firmware just fine.

```
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/dev/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/dev/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/dev/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/dev/FrameworkArduino/main.cpp.o
Compiling .pio/build/dev/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/dev/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/dev/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/dev/libFrameworkArduino.a
Indexing .pio/build/dev/libFrameworkArduino.a
Linking .pio/build/dev/firmware.elf
Building .pio/build/dev/firmware.bin
Retrieving maximum program size .pio/build/dev/firmware.elf
Checking size .pio/build/dev/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  27.8% (used 81912 bytes from 294912 bytes)
PROGRAM: [========= ]  88.4% (used 1737694 bytes from 1966080 bytes)
================================================================ [SUCCESS] Took 50.35 seconds ================================================================

Environment    Status    Duration
-------------  --------  ------------
prod           IGNORED
dev            SUCCESS   00:00:50.346
================================================================ 1 succeeded in 00:00:50.346 ================================================================

```

gepd, 2019-08-25:
Can you try to remove this folder  `/Users/srikrishnachaitanyanarumanchi/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv`
It seems like deviot installed platformio but as you aready have installed, it shouldn't be necessary

Restart ST after remove the folder

krish2487, 2019-08-26:
@gepd  

That worked !!! :-)
Thanks a ton!!

I am closing the issue now.

closed 2019-08-26 by krish2487