gepd/Deviot #296: Strange problems suddenly with uploading

tablatronix, 2019-11-17:
When I upload via serial it acts like it is doing ota, and just sits there, 
Cannot upload, it always fails as if its waiting for ota??

Serial monitor works fine.

```PHP
Creating BIN file ".pio/build/nodemcuv2_dev/firmware.bin" using ".pio/build/nodemcuv2_dev/firmware.elf"
<lambda>(["upload"], [".pio/build/nodemcuv2_dev/firmware.bin"])
MethodWrapper(["upload"], [".pio/build/nodemcuv2_dev/firmware.bin"])
Use manually specified: /dev/cu.SLAB_USBtoUART
"/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python" "/Users/shawn/.platformio/packages/tool-espotapy/espota.py" --debug --progress -i /dev/cu.SLAB_USBtoUART -f .pio/build/nodemcuv2_dev/firmware.bin
12:07:26 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '/dev/cu.SLAB_USBtoUART', 'host_port': 13344, 'image': '.pio/build/nodemcuv2_dev/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
12:07:26 [INFO]: Starting on 0.0.0.0:13344
12:07:26 [INFO]: Upload size: 378848
Sending invitation to /dev/cu.SLAB_USBtoUART ..........
12:09:07 [ERROR]: No response from the ESP
*** [upload] Error 1
```

Also get this error now and then testing this.
```PHP
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 241, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 372, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 184, in run_command
    if(self.cwd):
AttributeError: 'Command' object has no attribute 'cwd'
reloading settings Packages/User/deviot.sublime-settings
```

tablatronix, 2019-11-17:
it got stuck , something somewhere was overriding it, and thought I was trying to ota AND serial ?

I had to device, no connection to reset it.

gepd, 2019-11-22:
Are you using the dev version?

The OTA information is stored in `deviot.sublime-settings` located in `Packages/User` try removing the `auth_pass` option

tablatronix, 2019-11-22:
Dev, of course