gepd/Deviot #311: DEV Critical Bugs

tablatronix, 2020-04-22:
Logging critical bugs that must be fixed for deviot to be usable at all at this point.
( I assume no-one is still using this or has switched to vscode now )
If anyone wants to track these or work on them , please comment here or post your forks.
I will try to start once I have the codebase figured out and some pdb debugging setup,

**Enable logging in deviot sublime package settings logger_level**
_start_listener.py doesn't seem to work_

Most of these seem fairly simple state corruption, flags, or order of operation sync checks
Should be fixable fairly easily

## Platformio.ini corruption
The config reader or writer corrupts newlines, it does this if there is multiple white space between `var        = value` // eg.
becomes
`var = 
value
`
and pio chokes

There are some other syntax it chokes on also, some comment blocks, and some valid multi line env declarations, and indentation.

### workaround
disable all ini write flags TRUE, might have to fiddle to get environments to load though.

## cwd.run
```
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 241, in _finish
    run_next()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 372, in run_next
    Command().run_command(_COMMAND_QUEUE.popleft())
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 184, in run_command
    if(self.cwd):
```

## Going from OTA back to serial
Continues to assemble cmd with serial and ota ports in args

### workaround
Select devices-> Not Used, sometimes will clear it up

## Serial monitor state
Serial monitor is losing state, says its running, even if it cannot start, or runs and deviot ungreys serial options "send" and fails to disable it when uploading causing upload failures

tablatronix, 2020-05-11:
I just found something interesting when there is an ota port, serial monitor is not opening, well it opens but doesn't actually connect to serial port, says its running ( green bar )

If I kill all my ota devices on my network, it started working, also this shows up in my console everytime, which is where I got the hint, also noticed that ota ports were not showing in serial list, and they used to.  I might have to start reverting stuff and see where this all broke.

```
using gpu buffer for window
192.168.1.76
{'port': 8266, 'auth_upload': 'no', 'ssh_upload': 'no', 'board': '"PLATFORMIO_NODEMCUV2"', 'weight': 0, 'address': '192.168.1.76', 'priority': 0, 'tcp_check': 'no'}
192.168.1.76
{'port': 8266, 'auth_upload': 'no', 'ssh_upload': 'no', 'board': '"PLATFORMIO_NODEMCUV2"', 'weight': 0, 'address': '192.168.1.76', 'priority': 0, 'tcp_check': 'no'}
```

tablatronix, 2020-05-11:
So thats ok, I have to wait until it breaks and see what bad data is messing that up and exceptioning

tablatronix, 2020-05-11:
ok the second serial issue, serial starts up says it connected, gree bar, nothing in pane, If I goto send and send something, I get a huge dump of bad characters then serial output.

Looking into this next

tablatronix, 2020-05-19:
I found out that the serial port uploading as OTA, has to do with a damn platformio feature.

https://github.com/platformio/platform-espressif8266/commit/54cd0fb1e2f61b1c6624d708f749372a37dd9648

If the port resolves as a host, and some stupid damn routers resolve anything, it breaks, pio now adds port to boards.txt and expects you to pass that to pio args
`generic.upload.tool=esptool` so damn stupid.. autodetect works though.

Still not sure how its supposed to be fixed, as its should be getting it from target..

https://github.com/platformio/platform-espressif8266/issues/134

gepd, 2020-07-12:
I'm checking all this.

I've tried to reproduce the "Platformio.ini corruption" adding spacing to vars but I'm not getting the corruption. Can you provide an example to reproduce it?

Investigating... "Going from OTA back to serial"

(next time would be easy if you split it in multiples issues)

tablatronix, 2020-07-12:
Of course, this is a summary of issues I have already posted, I know you have not been active, So I thought it would help if I prioritized the major issues, so others can help and you can catch up if you want.

I spent a few days going over the 5 major bugs I have posted and tried to establish if they are related or a root cause or related dups.

I can link to the original issue here, but they are a old and stale, and even I might not be able to reproduce anymore.

tablatronix, 2020-07-12:
I linked #272 above, it is a library that uses platformio.ini, when I compile, deviot rewrites the ini file ( to insert baud and lib deps etc ) and corrupts the file. It does it with anything though (osx),  marlin etc