gepd/Deviot #313: Infinite recursion when searching for a library

ivankravets, 2020-07-09:
Hi @gepd,

Thanks again for this plugin! We see hundreds of thousands invalid requests to our registry API:

```
/lib/search?query=Adafruit+TouchScreen*page=50&query=Adafruit+TouchScreen"
```
1. Multiple `query` params
2. Invalid `page`, it should be `&page=...`

Could help with a fix for this issue? Otherwise, we will block all requests that come from `Deviot/x.y.z (Sublime-Text/N)`.

There are dozens requests per second from each developer machine.

ivankravets, 2020-07-09:
1. Please remove `*` from https://github.com/gepd/Deviot/blob/ce149bc2bffdc446bf37cb3c07a614d60fc51923/libraries/libraries.py#L96
2. Please fix https://github.com/gepd/Deviot/blob/ce149bc2bffdc446bf37cb3c07a614d60fc51923/libraries/libraries.py#L110

ivankravets, 2020-07-11:
Thanks 🙏 for the quick fix! There is the problem with concatenation in 110 line. Multiple “query” args. 

Also, could you limit up to 10-20 results? No need to fetch 10k libraries in loop. I think first 10-20 results are enough.

ivankravets, 2020-07-11:
Or , just 1 call and no need to go through all pages. To test this - please change http user agent to something that does not contain Deviot. We currently block all requests because they overload our servers. When user does simple search - it fetches thousands libraries in recursion without any delay which freezes Sublime. 

So, I think the 1 request and first 10 results are enough.

gepd, 2020-07-11:
> Thanks 🙏 for the quick fix! There is the problem with concatenation in 110 line. Multiple “query” args.

You mean this? https://github.com/gepd/Deviot/commit/0916c9e0883727becbbd5f0a26578a401cf52176

> Also, could you limit up to 10-20 results? No need to fetch 10k libraries in loop. I think first 10-20 results are enough.

Done https://github.com/gepd/Deviot/commit/3b81be5fc1bf8afd33dfd03ba93c5d8f76728ec8

closed 2020-07-11 by gepd

ivankravets, 2020-07-11:
Yes, now everything looks great! Thanks!