gepd/Deviot #314: Deviot overriding  build systems for C / C++ files

krish2487, 2020-07-22:
Hello.. This is not technically an issue per say. but it would be nice to not force the associations of C / cpp files with deviot build system. It is an amazing plugin, but I also work with other C projects and it is a pain to reset the syntax to C every time  I open a c / cpp file. I have other plugins like linters and LSP which work only if the build system is set to C or C++ depending on the extension. Is there a way to reset these associations in sublime?? Alternatively, can this be made into a menu item that can be turn on and off?

### Operating system

Mac OS 10.14.6 Mojave

### Board used

Not relevant.

### Error(s) in ST Console? (View > Show Console)

Not relevant

### Steps to Reproduce

Install deviot from package control and open any C / cpp project or file. The build system indicator on the bottom right defaults to deviot instead of C / C++. It can be reset explicitly to C or C++ but it is preferable to have the defaults to C / C++ and explicitly set it to deviot for a platformio project.

### Actual Results
NA

### Expected Results
NA

### Additional info
NA

rwols, 2020-07-23:
FWIW this was fixed in LSP by adding this syntax to the list of languages, something like this:

```
        {
          "languageId": "cpp",
          "scopes": [
            "source.c++",
            "source.iot"
          ],
          "syntaxes": [
            "Packages/C++/C++.sublime-syntax",
            "Packages/Deviot (Arduino IDE)/deviot.sublime-syntax",
          ]
        },
```

I would suggest to at least keep the base scope as `source.c++`, instead of `source.iot`. This way, your syntax will auto-work in ST4 which uses just these base scope selectors.

krish2487, 2020-07-23:
@rwols  The file syntax is cpp or c. if that is what you are suggesting..

krish2487, 2020-08-09:
Ok. For anyone else that might run into the same issue. 
This is how I hacked around it.

Comment out the lines ~ 42/43 in **User-Packages-Deviot (Arduino IDE)-Deviot.py**

```
   # Load or fix the right deviot syntax file 
     Syntax().paint_iot_views()
```

This prevents deviot from overwriting the syntax associations with C/C++/h files everytime sublime text starts. 
The flip side is that on an actual deviot project, the syntax has to be manually set to deviot.

I am closing the issue now.

closed 2020-08-09 by krish2487

timsky, 2020-11-25:
> The flip side is that on an actual deviot project, the syntax has to be manually set to deviot.

In file "tools.py" (Data\Packages\Deviot (Arduino IDE)\libraries\) on line 25.
I left just one extension there:
`accepted = ['ino']`