gepd/Deviot #320: Support for python3

hawaiik, 2021-01-16:
As platformio dropped support for python 2 & 3.5 how can we update deviot to python 3?https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

atwinny, 2021-02-23:
Any suggestions?

berendkleinhaneveld, 2021-02-24:
I also got this message and an error when I tried to compile a cpp file. I removed the Deviot plugin and tried to reinstall the plugin. Then I saw a message telling me that `/usr/local/opt/python/bin/python3.7` couldn't be found.
I'm on a mac, so I did the following to install python3.7:

`brew install python@3.7`

This installs python to `/usr/local/opt/python@3.7/bin/python3`. Then to make Deviot find python 3.7, I just made a symlink at the path from the error message:

`mkdir -p /usr/local/opt/python/bin
ln -s /usr/local/opt/python@3.7/bin/python3 /usr/local/opt/python/bin/python3.7`

After doing all that I installed Deviot again and now things seem to be working fine. Hope this helps.

tablatronix, 2021-05-12:
Really? so you are using global enviroment?

tablatronix, 2021-05-12:
deviot settings 
`env_path`

tablatronix, 2021-05-12:
ok well I could not get those to work, if deviot penv is already installed its a pita..

Easiest thing for me was to 

OSX
Remove all platformio
Install platformio CLI ( via pio instruction scripts )

Remove deviot penv, remove deviot.ini
restart, let deviot install check env paths and python on its own and use global penv

We will see if it has issues.

hijomojo, 2021-09-22:
I know it's been months, but for anyone still wondering (who hasn't given up 😛) my guide above should help you get things working again.