gepd/Deviot #323: As platformio now requires python 3.x while Deviot requires 2.x, is Deviot no more ? Any work-around ?

fiveangle, 2021-02-09:
Thanks !

-=dave

timsky, 2021-02-25:
> Winpython32-3.8.7.0dot.exe:
`Python 3.8.7 (tags/v3.8.7:6503f05, Dec 21 2020, 17:43:54) [MSC v.1928 32 bit (Intel)] on win32`
and
`Deviot 2.3.0.dev13`
Works fine!

oopsShouldntHaveDoneThat, 2021-03-10:
> 
> 
> **Winpython**32-3.8.7.0dot.exe:
> `Python 3.8.7 (tags/v3.8.7:6503f05, Dec 21 2020, 17:43:54) [MSC v.1928 32 bit (Intel)] on win32`
> and
> `Deviot 2.3.0.dev13`
> 
> Windows 7 x64. Works fine!

Are you willing? This goes over my head.

hijomojo, 2021-09-22:
Pretty sure I've figured out a relatively easy solution for you all. Take a look at the guide "issue" I made in the mention link above this comment.

Kreativnoime123, 2023-03-01:
doing all of this correctly in my mind, but it doesn't work for me, when I go to upload it says select from list and no list pops up