gepd/Deviot #329: ST4 install library fail

tablatronix, 2021-06-22:
```
Exception in thread Thread-35:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/alverson/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 126, in download_list
    quick_panel(self.quick_list, self.library_install_async)
  File "/Users/alverson/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_panel.py", line 19, in quick_panel
    window.show_quick_panel(items, callback, flags, index)
  File "/Applications/Sublime Text.app/Contents/MacOS/Lib/python33/sublime.py", line 728, in show_quick_panel
    item_tuples.append((item[0], "\x1f".join(item[1:])))
TypeError: sequence item 1: expected str instance, int found
```

tuple expects string, ID is int

fix?

```CPP
    def quicked(self, source_list):
        """Quick panel List
...
            quick_list.append([info, description, str(id)])

        self.quick_list = quick_list
```

tablatronix, 2021-10-02:
another issue

```
generating syntax summary
Exception in thread Thread-17:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 120, in download_list
    self.quicked(response_list['items'])
  File "/Users/shawn/Library/Application Support/Sublime Text/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 143, in quicked
    frameworks += framework + ' '
TypeError: unsupported operand type(s) for +: 'dict' and 'str'
```

tablatronix, 2022-02-23:
In ST4 there seems to be an issue with passing in the pagination items

siddhpant, 2022-11-04:
I changed the offending line in `quick_panel.py` to below and it worked:

```python
    window.show_quick_panel([[str(i) for i in item] for item in items],
                            callback, flags, index)
```

tablatronix, 2022-11-04:
Thanks, I have not PR anything because this project is dead.

here is my fork, i try to submit hacks, I need to properly fix some stuff when i have time, some stuff is just turned off..

https://github.com/tablatronix/Deviot/commits/st4

siddhpant, 2022-11-04:
> Thanks, I have not PR anything because this project is dead. But I post issues here for posterity
> 
> here is my fork, i try to submit hacks, I need to properly fix some stuff when i have time, some stuff is just turned off..
> 
> https://github.com/tablatronix/Deviot/commits/st4

Thanks. 

Maybe you should rebase on the develop branch instead of branching from master. master (and hence your branch) is 157 necrocommits behind.