gepd/Deviot #330: Guide: Running Deviot with ST3 / ST4, PlatformIO 5, and Python 3.6+

hijomojo, 2021-09-22:
Hi all,

Just figured I'd post this here as a solution to potentially many of your problems with running Deviot since PIO ended support for python 2.7/3.5 and lower. I've used these steps on ST3 and ST4 for Windows 10, and ST4 for macOS 11(.6)

**For simplicity's sake, start from scratch** 
- Remove Deviot from Sublime if you have it installed already (Package Control > Remove Package > Deviot)
- If you are on mac and have/had the dev branch installed, unfortunately I don't think it works with these instructions. If someone figures out a solution to that then let us know here! Removal is of course: Package Control > Remove Repository > https://rawgit.com/gepd/Deviot/master/repository.json
- If you're on mac and have installed PlatformIO via their get-platformio.py script, my understanding is you'll need to remove it. I had to use the Homebrew install, and I'm not sure how this goes if you've already got the regular install in place.
`pip uninstall platformio`

**Next step, Python**
- If you have 2.7 and 3.6+ installed already, you may need to re-install your 3.6+ version to make sure it has precedence over 2.7 (I'm not really that familiar with python so this may not strictly be necessary)
- If you don't have Python 2.7 and/or 3.6+ installed, install them however you like (e.g. via [](https://www.python.org/downloads/)). As above, make sure 3.6+ is installed/reinstalled after 2.7. My understanding is that you do need both versions, but please correct me if I'm wrong, it's always nice to minimise install overheads. In #320 and #323 people have mentioned they have it working in mac/Windows without 2.7, I haven't tested scientifically enough.

**Then on to PlatformIO**
macOS
- Install platformio via Homebrew. This seems to be important, I could not get it to work by installing PIO via their get-platformio.py script.
- If you don't have Homebrew already, it can be had by running (in terminal):
- `/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"`
- If it requests you install the Command Line Tools for Xcode, go ahead with that
- Then install platformio:
- `brew install platformio`

Windows
- It's been quite a few months since I did this, but IIRC you may be able to get away with the PIO install created by Deviot. Check if you have a .platformio folder in your user directory e.g. C:\Users\*username*\.platformio. If not...
- Install PIO using their instructions here: [](https://docs.platformio.org/en/latest/core/installation.html#id30)

**Both**
- Install Shell commands, a.k.a Set PATH variables, for PIO. Follow PIO's instructions here: [](https://docs.platformio.org/en/latest//core/installation.html#install-shell-commands)

**Finally, back in Sublime Text**
- Install Deviot: Package Control > Install Package > Deviot

**Optional: Windows Only**
- Switch to Dev branch for most up to date version of Deviot. Instructions here: [](https://github.com/gepd/Deviot/wiki/Setup#developer)
As I mentioned above, this seems to break the mac install such that the Deviot menu disappears even though Package Control says it's installed.

Hope this all helps!

hijomojo, 2021-09-22:
This process also solved various issues for me similar or identical to #318, #328, #324, #326, #268

OneALab, 2021-11-25:
It looks to me like there are three instances in the above guide that make references to instructions that are not provided.

    "Install PIO using their instructions here:" 
    "Install Shell commands, a.k.a Set PATH variables, for PIO. Follow PIO's instructions here:"
    "Switch to Dev branch for most up to date version of Deviot. Instructions here:"

Can anyone fill in the blanks? 

Thanks.

hijomojo, 2021-11-25:
Oh, thanks for spotting that!
Looks like I missed putting the plain text in for the link markdown...the URLs where there (at least in the post editor), but not visible.
I've fixed the original post now

hijomojo, 2021-12-19:
Very welcome!

I think you're right about the PATH stuff though....by chance I'm playing around with an RP2040 board at the moment which required installing the raspberrypi platform, and during that process I got what seemed to be an error. Long story short I tried to run `pio` in the Windows terminal and found I didn't even have PATH set in order for that to work... I recently updated ST3 to ST4 on this computer, expecting to need to come back to my guide here, but Deviot ran without issue following the update, and has done ever since.

**Extra info maybe useful for future reference**
I found after I set PATH to the platformio install in my user directory, that the version there did not match the version reported by the same `pio --version` command in Deviot's pio terminal. This is because, for some reason, Deviot has its own copy of the `..\penv\Scripts` folder in the ST packages folder in my appdata folder (i.e. `C:\Users\Username\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv`). I have no idea why this is here, there isn't a complete pio install located there (e.g. platform and package files are still in `C:\Users\username\.platformio`), and my other computer does not have a `\penv` folder in either the original ST3 appdata folder, or the newer ST4 one. On that computer I installed ST4 next to ST3, whereas on this one I just updated over the top (apparently this means keeping the old folder's names untouched). Not sure what's going on, but it seems everything works for me...and if the guide still works I'm not fussed 😀

hijomojo, 2022-01-09:
@QyuriLa 
> It seems that it happens on Windows (or at least my PC) too.
> ... I ended up staying in the stable release (2.2.2), though.

I actually just ran into this issue when upgrading one of my Windows machines from stable to dev version 2.3.0.dev13, too!
I've just figured out what happened and how to fix it ... basically, the dev installer fails to create a few files but you can manually replace them. Here's a link to my solution:
#337

hijomojo, 2022-01-09:
#337

gianlucaf81, 2022-06-29:
OMG! after years, with this guide, I was able to get it working again. thx

siddhpant, 2022-11-04:
This writeup seems to refer to itself in place of other links.

siddhpant, 2022-11-04:
Also, dev branch can be installed by siumply adding link to GitHub repo instead of the rawgit link.

hijomojo, 2022-11-27:
> This writeup refers to itself in place of other links.

Fixed that, I had linked issue 330 instead of issue 337, sorry.

> Also, dev branch can be installed by simply adding link to GitHub repo in package control instead of the rawgit link.

I'll take your word on that, I just figured using the process gepd wrote would keep the info more consistent. Wouldn't say one copy and pasting one link is more simple than another.

jon-daemon, 2024-10-04:
On a fresh install of MacOS 13 with ST4 I tried to install Deviot.

I installed platformio with homebrew like in this guide. Also I updated python3 to python3.12.
Homebrew created a symlink `python` to `python3.12` in `/usr/local/opt/python@3.12/libexec/bin`.
I added `/usr/local/opt/python@3.12/libexec/bin` in my $PATH so `python3` could also be used as `python`.

I tried Deviot but I was still getting a popup for installing python2. I created a symlink `python2` to `python3` in `/usr/local/bin/` and Deviot stopped asking for python2. So it was searching for python2 specifically.

I changed "python2" to "python3" in `Deviot (Arduino IDE)/libraries/tools.py` and it stopped showing the popup for python2 installation. Is this enough. I was able to compile and upload a simple sketch.

hijomojo, 2024-10-05:
I don't use macOS day to day so I don't know why you might have needed those extra steps compared to my guide, but I guess it's ok if it works for you! Did you end up removing the python2 > python3 symlink or did you need to keep that in conjunction with the edits in tools.py? If you kept it, would that not potentially cause issues with other software if it needs python 2.x?

Re "Is this enough?"...You could test some other features like the serial monitor or uploading to a board which doesn't use Serial for upload (like Teensy, if you have any), and see if you run into any issues.

jon-daemon, 2024-10-06:
Your guide was working on my older MacOS installation where both python2 and python3 were installed. Now MacOS has only python3 installed.
I deleted the symlink python2>python3 after editing the tools.py file. I tested the serial monitor and uploading with a usbasp and they work fine.

jon-daemon, 2024-11-15:
I installed it on a new Mac computer with OS X 15.0 and Sublime Text Build 4169. I found that editing `tools.py` I mentioned above is not correct. It was working because I had created symlinks python > python3 prior to that and Deviot required to see that only during the first run. So creating the symlink and deleting them was enough.

This is what worked for me on my new installtion:
- Installed platformio using homebrew
- Added the dev branch repo before installing Deviot (for some reason it was not updating to dev branch if it was already isntalled)
- Perform one of the following fixes for python3:
    A. Create a python > python3 symlink.
    B. Add a dir in PATH containing unversioned python. I installed python3.13 so I could add `/opt/homebrew/opt/python@3.13/libexec/bin` in PATH where there is already a python symlink pointing to python3
    C. Edit `Deviot (Arduino IDE)/beginning/check_requirements.py` and change `_symlink = 'python2'` to `_symlink = 'python3'`

I used the third method and it seems to work. The only issue I have on my new installation is that the port list doesn't  show the connected device. That could be an issue with the new computer which has only USB 3 input ports and the device is connected using a USB type C to type A  adapter. 

I selected `Not Used` and it could still auto detect it but I'd like to be able to see the available ports
```
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: /dev/cu.usbserial-2110
```