gepd/Deviot #331: OSX: Serial ports not showing on big Sur

tablatronix, 2021-10-03:
For some reason on my new computer I cannot get serial ports to show i the drop down..
Pio sees them, perhaps deviot uses some generic python module to do this instead of using pio devices list.. maybe a perm issue?

tablatronix, 2021-10-03:
Ok so only .cu are being seen , no tty. there is no address so they are filtered

```
/dev/cu.wlan-debug
n/a
n/a
/dev/cu.debug-console
n/a
n/a
/dev/cu.Bluetooth-Incoming-Port
n/a
n/a
/dev/cu.usbserial-0170E926
n/a
n/a
/dev/cu.SLAB_USBtoUART
n/a
n/a
```

tablatronix, 2021-10-03:
workaround for now was to just ignore it, so i can at least use them, not sure why these are missing yet, probably module outdated

```
        for dev_name in dev_names:
            if(address != 'n/a' and dev_name in port_no or platform() == 'windows'):
            if(dev_name in port_no or platform() == 'windows'):
                serial_ports.append([description, port_no])
```

jon-daemon, 2024-11-15:
Your fix is working. Have you found a solution to show the description in the drop down list?