gepd/Deviot #336: can't see devio menu

laomatt, 2021-12-12:
Before report an issue, please make sure it hasn't been reported before.

If you need information about an option or feature make sure to first look into the [Deviot Wiki](https://github.com/gepd/Deviot/wiki)

### Operating system
macOS big sur v 11.2.3
### Board used
elegoo mega 2560 r3
### Description of problem
I cannot get 'deviot' to show up in my top menu on sublime text, i have removed and installed it many times
### Error(s) in ST Console? (View > Show Console)
ignored packages updated to: ["Deviot (Arduino IDE)", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading settings Packages/Deviot (Arduino IDE)/deviot.sublime-settings
reloading python 3.3 plugin Deviot (Arduino IDE).start_listener
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
generating meta info summary
generating syntax summary
reloading settings Packages/User/Preferences.sublime-settings
Unable to open /Users/matt/Library/Application Support/Sublime Text 3/Packages/Default/Preferences.sublime-settings
Unable to open /Users/matt/Library/Application Support/Sublime Text 3/Packages/Default/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
ignored packages updated to: ["Vintage", "Deviot (Arduino IDE)"]
reloading settings Packages/Deviot (Arduino IDE)/deviot.sublime-settings
unloading python 3.3 plugin Deviot (Arduino IDE).start_listener
generating syntax summary
generating meta info summary
### Steps to Reproduce

### Actual Results
can't see deviot menu
### Expected Results
should see a deviot menu
### Additional info

*You can edit the text above according to your need*

jaireaux, 2021-12-20:
I'm having the same problem but my Console has a different error.

```
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/Lib/python33/sublime_plugin.py", line 400, in load_module
    m.plugin_loaded()
  File "/Users/jaireaux/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/Deviot.py", line 34, in plugin_loaded
    PioInstall()
  File "/Users/jaireaux/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/beginning/pio_install.py", line 62, in __init__
    beginning_check()
  File "/Users/jaireaux/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/beginning/pio_install.py", line 291, in beginning_check
    save_sysetting('telemetry_check', True)
  File "/Users/jaireaux/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/tools.py", line 175, in save_sysetting
    def get_setting(key, default=None):
FileNotFoundError: [Errno 2] No such file or directory: '/Users/jaireaux/Library/Application Support/Sublime Text 3/Packages/User/Deviot/deviot.ini'
generating syntax summary
generating meta info summary
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
```

hijomojo, 2022-01-09:
This is a bit outside my knowledge to say for sure, but what it looks like to me is that Python is having an issue running those scripts, so Deviot is not actually being properly installed, resulting in Sublime throwing errors when trying to load it.
My understanding is that Deviot requires Python 2.7, do you have that installed?
You'll of course need Python 3.6+ to run Platformio within Deviot, so AFAIK you actually need both 2.7 and 3.6+ installed, and from my experience 3.6+ should be installed after 2.7

Judging by your paths, you're running ST3, right?

jaireaux, 2022-01-10:
Yes, I'm running ST3. I have two versions of Python installed.

```
(base) jaireaux@JROMBPRO /usr/bin % python2.7 --version
Python 2.7.18
(base) jaireaux@JROMBPRO /usr/bin % python --version
Python 3.9.9
```

I don't know how to force ST3's Package manager to use 2.7.

hijomojo, 2022-01-12:
@jaireaux You shouldn't need to force either one...at least I haven't had to do so on any of my machines, mac or win. I didn't read properly before, but I now see the error is with that .ini file apparently not existing. This is really beyond my knowledge of Deviot now though...best I can suggest is checking permissions for the folders or looking into whether some other macOS permissions thing is playing prison guard on the python/ST processes and preventing them from writing there.

@laomatt do you have both python 3.6+ and 2.7 installed? Running Sublime Text 3 or 4? And, are you running the regular/stable release or the dev branch of Deviot? If you're trying to run the dev branch and Deviot is actually working (e.g. you can use the keyboard shortcuts to do things like open the boards menu and compile things), then try my solution to the missing menu issue in #337