gepd/Deviot #337: Guide: Fix Deviot menu & syntax highlighting missing

hijomojo, 2022-01-09:
Hi All,

You might have run into an issue with Deviot (especially after installing the dev version in Sublime Text 4) where the Deviot menu doesn't show up, and syntax highlighting doesn't work and isn't available to pick from the syntax list.
I've just figured out how to fix this!

What's happened is that for some reason the dev branch install process fails to create a few files which relate to these features. It seems these are:
1. Context.sublime-menu
2. Default.sublime-commands
3. deviot.sublime-completions
4. deviot.sublime-syntax
5. Main.sublime-menu

If you've installed Sublime Text 4 (ST4) alongside ST3, then you can copy those files from your old Deviot install (if you had one) to the ST4 Deviot folder. This is:
macOS: `/Users/*username*/Library/Application Support/Sublime Text/Packages/Deviot (Arduino IDE)`
Windows: `C:\Users\*username*\AppData\Roaming\Sublime Text\Packages\Deviot (Arduino IDE)`

If you're using ST3 or have upgraded ST3 to ST4 directly, then the folder will be:
macOS: `/Users/*username*/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)`
Windows: `C:\Users\*username*\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)`

If you don't have those files, here's a copy of my ones from a relatively fresh environment of Deviot 2.3.0.dev13:
[deviot 2.3.0.dev13 menu files minimal.zip](https://github.com/gepd/Deviot/files/7834353/deviot.2.3.0.dev13.menu.files.minimal.zip)
It seems that the '-completions' and '-syntax' files get dynamically updated as you use Deviot; the ones I've uploaded here are relatively minimal.

I believe this solution will only work if Deviot is otherwise properly installed and functional. For example, in my installs, I could use the hotkeys to do things like select environment/board/port, and I could see this at the bottom of the ST window:
[image]
The board and port will show as " - " for you if you haven't selected them yet, but as long as "Deviot 2.3.0.dev13" shows up there when you have a compatible file open (e.g. a .ino file) the solution above should work.

I've tested this with Windows 10 and macOS 11.6

hijomojo, 2022-01-09:
If you're having issues with ST4 in general, here's an install guide which might help: #330

siddhpant, 2022-11-04:
I already have the files after fresh installation, so this isn't really a solution. I am on ST4.